InfoAmazonia/rede-site #56: home page online shows 20 equipments

vjpixel, 2015-10-04:
right now we have 29 equipments (2 prototypes), and at some moment is possible to have 24 turned on at the same time.

can you set the home page to show an unlimited list?

miguelpeixe, 2015-10-06:
that's on the api

vjpixel, 2015-10-07:
I still see the same problem

vgeorge, 2015-10-07:
Because it wasn't been deployed, it's online now. 

We are not deploying after every commit, so please don't reopen issues that were solved for code available at this repo. :wink:

closed 2015-10-07 by vgeorge

vjpixel, 2015-10-07:
so I ask to please only mark as done when I can check it. we're on the
final week before presenting this site in Santarém, so I need to give you
feedback in a way that we have the site done until Tuesday.

it's ok or will break your work procedure?

commit 742e72fc (2015-10-07): fix #56 by setting default sensors per page to 50