InfoAmazonia/rede-site #59: why there are 3 colos of lines?

vjpixel, 2015-10-07:
on the sensors' page [1], why three different colors of lines?

[1] http://rede.infoamazonia.org/sensors/560ee58e860f948069897df4/

miguelpeixe, 2015-10-07:
lowest, average and highest grades

miguelpeixe, 2015-10-13:
Apply this graph instead https://developers.google.com/chart/interactive/docs/gallery/intervals#bar-intervals

vjpixel, 2015-10-14:
when I put the mouse over the point, it's not clear what number is the maximum e what's the minimum.

jywarren, 2015-10-19:
Hi, all - any interest in collaborating on parts of this infrastructure? We've started work on a d3.js based graphing system for sensor data, called Grapherate: https://github.com/jywarren/grapherate -- It could be swapped for Google Charts, and provide more customization and be open source.

miguelpeixe, 2015-11-04:
Hi @jywarren, thank you for the suggestion. We would love to integrate an open source solution for the charts. I thought of making it in d3 myself but it would take more time than I have now. I looked at your solution and couldn't find how it handles `Date` values on the axis label display and how interactivity works (specially for aggregates, that combines min, max and average data).

closed 2015-11-12 by vjpixel