InfoAmazonia/rede-site #71: Need "Access-Control-Allow-Origin" response header to be "*" for external javascript access of data

jywarren, 2015-10-22:
Following up on [this discussion](https://groups.google.com/forum/#!topic/plots-dev/YpX-6UC5H8E) where we're having trouble loading your data, it looks like you can just make your response header for "Access-Control-Allow-Origin" to be "*" (wildcard)-- without that, javascript won't be able to access it from another domain. Vitor, would you be able to do that? 

The error I currently get is:

> XMLHttpRequest cannot load http://rede.infoamazonia.org/api/v1/sensors/560ee510860f948069897de3/measurements/ph.csv. No 'Access-Control-Allow-Origin' header is present on the requested resource. Origin 'http://jywarren.github.io' is therefore not allowed access.

This will enable [Grapherate](https://github.com/jywarren/grapherate) to read rede.infoamazonia.org data feeds!

vgeorge, 2015-10-27:
Hey @jywarren, we've just deployed a fix for that. Let me know if it is working for you.

jywarren, 2015-10-27:
It works!

http://jywarren.github.io/grapherate/g/?url=http://rede.infoamazonia.org/api/v1/sensors/560ee510860f948069897de3/measurements/ph.csv

closed 2015-11-03 by vgeorge