InfoAmazonia/rede-site #96: Erro ao enviar leituras ao servidor

virgiliol, 2018-07-03:
Foi configurado no Arduino o id +5593991162803 e configurado um novo dispositivo na plataforma InfoAmazônia com o Id referente e nome de ICTA UFOPA como mostra os prints e tentamos enviar as leituras para o servidor. Segue em anexo os prints dos erros. 
Questionamentos:
Todos os sensores precisam estar conectados para uma postagem na plataforma com sucesso?
Existe algum tipo de restrição no banco de dados da plataforma para upload das leituras?
De onde possivelmente devem ser esses erros? 

[image]
[image]
[image]

vgeorge, 2018-07-03:
@virgiliol 

O parâmetro `data` está mal-formado. O primeiro item deve ser uma data no formato ISO, no README tem um examplo. Outra coisa, o sensor está cadastrado no painel de controle do rede? É preciso cadastrá-lo antes dos envios. Sobre suas perguntas:

#### P: Todos os sensores precisam estar conectados para uma postagem na plataforma com sucesso?

Não é necessário enviar todos os parâmetros, mas o *request* deve conter pelo menos um.

#### P: Existe algum tipo de restrição no banco de dados da plataforma para upload das leituras?

Não entendi a pergunta. Para um envio ser aceito ele deve estar bem formatado e o sensor pré-cadastrado.

#### P: De onde possivelmente devem ser esses erros?

O *request* retorna uma matriz com os erros. Não conheço o Arduino para te indicador como fazer, mas provavelmente a biblioteca que você está usando deve retornar o "response body", e aí você pode debugar por lá.

Me avise se conseguir fazer o envio, por favor.

virgiliol, 2018-07-04:
@vgeorge 

Achei a correção da formatação do parâmetro `data` era apenas uma linha de código que estava comentada que adiciona o char `0` ao `buffer`. 
Após essa correção retornei a tentativa de envios porém ainda sem sucesso obtive um status diferente dessa vez (segue na imagem).
[image]

vgeorge, 2018-07-04:
Este sensor está com id diferente, ele está cadastrado na plataforma? 

A resposta não contém um status HTTP válido, é preciso que seja impresso o conteúdo completo da resposta do servidor.

vgeorge, 2018-08-19:
@virgiliol Como está esta questão? O problema foi resolvido?