InfoAmazonia/rede-site #98: Atualização da documentação

vgeorge, 2018-08-19:
Fiz uma revisão do README. Separei a [referência da API](https://github.com/InfoAmazonia/rede-site/blob/master/API-description.md) e a [documentação de desenvolvimento](https://github.com/InfoAmazonia/rede-site/blob/master/getting-started-pt.md) em arquivos à parte, pois estavam muito extensos. Abri esta issue para receber comentários sobre as alterações. 

Peço que algum desenvolvedor, se possível, use a documentação para confirmar se passos estão claros e permitem iniciar um servidor localmente sem problemas.

virgiliol, 2018-08-22:
Infelizmente após essa atualização não consegui mais iniciar o servidor. Tentei seguir os novos passos e quando tento iniciar a instância de desenvolvimento do MongoDB é retornado o seguinte erro:  client version 1.22 is too old. Minimum supported API version is 1.24, please upgrade your client to a newer version.

vgeorge, 2018-08-22:
Fiz uma busca sobre o erro e parece ser associado a problemas da versão do Docker para o Windows, não é um problema do RIA, a princípio. Fiz uma modificação no docker-compose.dev.yml, verifique se o seu Docker está atualizado e tente com tente novamente com a última versão deste repositório.

virgiliol, 2018-08-22:
@vgeorge consegui iniciar o MongoDB porém na instalação o npm retorna o seguinte erro: 
info cache git://github.com/miguelpeixe/angular-pickadate into angular-pickadate
Já existe uma subpasta ou um arquivo -p.
Erro ao processar: -p.
Já existe uma subpasta ou um arquivo public.
Erro ao processar: public.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! infoamazonia-rede@0.1.0 postinstall: `napa && mkdir -p public && ./node_modules/.bin/grunt build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the infoamazonia-rede@0.1.0 postinstall script.

vgeorge, 2018-08-26:
As tarefas do npm estavam configuradas para rodar no Linux, modifiquei-as agora para rodar também no Windows. Faça um teste, por favor.

virgiliol, 2018-08-27:
@vgeorge tudo certo. Tive apenas que usar o comando `--force-recreate` na hora de utilizar o docker para recriar a rede pois informava que não estava sendo encontrada.

vgeorge, 2018-08-27:
Ótimo!

closed 2018-08-27 by vgeorge

virgiliol, 2018-08-28:
@vgeorge a autenticação é feita por token? E esse token seria o id? Tem como recuperar esse id? 
Essas dúvidas surgiram ao trabalhar com a API

vgeorge, 2018-08-28:
Olha, que eu me lembre o endpoint do sensor não precisa de autenticação, apenas que o telefone informado esteja cadastrado na plafaforma. Mas é preciso confirmar lendo o código. De qualquer maneira, a API usa o módulo passport para autenticação dos chamados do Web App:

https://github.com/InfoAmazonia/rede-site/blob/master/package.json#L71-L73

virgiliol, 2018-08-28:
Estou tentando fazer o seguinte post '{"identifier":"+5593984038700","name":"teste","description":"","image":""}' e me é retornado :
'{
    "messages": [
        {
            "status": "error",
            "text": "access_token.unauthorized"
        }
    ]
}'